PRBonn/kiss-slam #6: Poses Estimation (Question)

harisbinyousaf24, 2025-03-19:
Hi creators @tizianoGuadagnino and others, thanks for creating such a cool SLAM algorithm (as i already foresaw the potential in KISS-ICP 2 years back and integrated IMU and GPS with that pipeline to give me global localized trajectory)

But here my question is related to poses estimation; Likely kiss-icp is used in backend for lidar odometry pose estimation and on top of that pose graph optimization happens but when i estimated my odometry poses using kiss ICP and kiss SLAM both give same poses which to my intuition was that KISS-SLAM might give better poses (my drive was in a straight line - no loop closures)..**does pose graph optimization happens in regular pose estimation or only when loop closures are detected?**

tizianoGuadagnino, 2025-03-20:
Hi @harisbinyousaf24, thanks for having a look at our project. By default, pose graph optimization happens just at the time of a loop closure.

Did you get different results between kiss-ICP and Kiss-SLAM without loop closure? It is interesting because I didn't expect that to happen. The only difference between the two is that Kiss-SLAM is constantly switching the local map in which it computes the odometry while kiss-ICP use always the same. Can you conferm that you get different poses in these cases?

tizianoGuadagnino, 2025-03-21:
Correct, until you don't have a loop that two should give you almost identical results. Let me know when you get more results

tizianoGuadagnino, 2025-03-26:
This is what I call the "pitch drift" issue of LiDAR odometry pipeline. I honestly haven't figure out why this happens, it is actually a very good research question to answer. I have my guesses but no numbers to support them.

To answer your question, yes the loop closure will just redistribute that z error across the trajectory, so you are simply gonna see it less but it will still be there.

harisbinyousaf24, 2025-03-26:
@tizianoGuadagnino I also dont know the exact cause of this issue but in my own way i may have found the temporary solution to this problem back then. The solution was not the one i would say is perfect or reliable but if i filter my pcd frames on z axis (Upwards) (keeping half the points i that axis - removing tree branches eg in urban scenarios) then this behaviour isnt noticeable atleast upto 1km drives. 

[image]

Side View:

[image]

But the whole point is if i cant get the local trajectory correct then how can i rely on SLAM solution ?

One way i thought to tackle this problem is may be on general ICP run , match the current frame with local accumulated map (especially ground surface features) this would give the offset that can be subtracted to keep the local poses accurate.

marmik18, 2025-04-01:
Sorry for a dumb question, what are you using to visualize the output from the `slam_output` folder?

harisbinyousaf24, 2025-04-02:
@marmik18 You mean the visualizer i am using for the above pics i shared? I actually use the pose file (npy) and pcds frames to stitch them on their poses to create full registered map and then used pcl_viewer.

benemer, 2025-04-23:
I will close this issue. Feel free to keep discussing or open a new issue :)

closed 2025-04-23 by benemer