PRBonn/kiss-slam #15: MemoryError: std::bad_alloc

darshanAtSwaayatt, 2025-04-02:
Trying to guess how to read your data: `pip install "kiss-icp[all]"` is required
[Open3D WARNING] The number of points is 0 when creating axis-aligned bounding box.
RegistrationVisualizer initialized. Press:
	[SPACE] to pause/start
	  [ESC] to exit
	    [N] to step
	    [C] to center the viewpoint
	    [W] to toggle a white background
	    [B] to toggle a black background

 63%|████████████████████▎           | 14375/22688 [56:36<32:44,  4.23 frames/s]
╭───────────────────── Traceback (most recent call last) ──────────────────────╮
│ /home/swaayatt7/.local/lib/python3.8/site-packages/kiss_slam/tools/cli.py:14 │
│ 2 in kiss_slam                                                               │
│                                                                              │
│   139 │                                                                      │
│   140 │   from kiss_slam.pipeline import SlamPipeline                        │
│   141 │                                                                      │
│ ❱ 142 │   SlamPipeline(                                                      │
│   143 │   │   dataset=dataset_factory(                                       │
│   144 │   │   │   dataloader=dataloader,                                     │
│   145 │   │   │   data_dir=data,                                             │
│                                                                              │
│ ╭──────────────────────────── locals ────────────────────────────╮           │
│ │          config = None                                         │           │
│ │            data = PosixPath('cloud_VLP32C')                    │           │
│ │      dataloader = 'generic'                                    │           │
│ │ dataset_factory = <function dataset_factory at 0x7ff2d98e38b0> │           │
│ │            jump = 0                                            │           │
│ │            meta = None                                         │           │
│ │         n_scans = -1                                           │           │
│ │    refuse_scans = False                                        │           │
│ │        sequence = None                                         │           │
│ │    SlamPipeline = <class 'kiss_slam.pipeline.SlamPipeline'>    │           │
│ │           topic = None                                         │           │
│ │       visualize = True                                         │           │
│ ╰────────────────────────────────────────────────────────────────╯           │
│                                                                              │
│ /home/swaayatt7/.local/lib/python3.8/site-packages/kiss_slam/pipeline.py:58  │
│ in run                                                                       │
│                                                                              │
│    55 │   │   self.refuse_scans = refuse_scans                               │
│    56 │                                                                      │
│    57 │   def run(self):                                                     │
│ ❱  58 │   │   self._run_pipeline()                                           │
│    59 │   │   self._run_evaluation()                                         │
│    60 │   │   self._evaluate_closures()                                      │
│    61 │   │   self._create_output_dir()                                      │
│                                                                              │
│ ╭───────────────────────────── locals ──────────────────────────────╮        │
│ │ self = <kiss_slam.pipeline.SlamPipeline object at 0x7ff2da39df10> │        │
│ ╰───────────────────────────────────────────────────────────────────╯        │
│                                                                              │
│ /home/swaayatt7/.local/lib/python3.8/site-packages/kiss_slam/pipeline.py:78  │
│ in _run_pipeline                                                             │
│                                                                              │
│    75 │   │   │   start_time = time.perf_counter_ns()                        │
│    76 │   │   │   self.kiss_slam.process_scan(scan, timestamps)              │
│    77 │   │   │   self.times[idx - self._first] = time.perf_counter_ns() - s │
│ ❱  78 │   │   │   self.visualizer.update(self.kiss_slam)                     │
│    79 │   │   self.kiss_slam.generate_new_node()                             │
│    80 │   │   self.kiss_slam.local_map_graph.erase_last_local_map()          │
│    81 │   │   self.poses, self.pose_graph = self.kiss_slam.fine_grained_opti │
│                                                                              │
│ ╭──────────────────────────────── locals ─────────────────────────────────╮  │
│ │        idx = 14375                                                      │  │
│ │       scan = array([[ 0.05218482,  2.12011814, -0.98892671],            │  │
│ │              │      [-1.22904372, 16.77647781, -0.29361928],            │  │
│ │              │      [ 0.42856497, 17.41135216, -0.50687349],            │  │
│ │              │      ...,                                                │  │
[... 35 lines cut ...]
│                                                                              │
│ ╭───────────────────────────────── locals ─────────────────────────────────╮ │
│ │ current_node = <kiss_slam.local_map_graph.LocalMap object at             │ │
│ │                0x7ff1ff292580>                                           │ │
│ │         self = <kiss_slam.tools.visualizer.RegistrationVisualizer object │ │
│ │                at 0x7ff1ffbe0f70>                                        │ │
│ │         slam = <kiss_slam.slam.KissSLAM object at 0x7ff1ffc6da30>        │ │
│ ╰──────────────────────────────────────────────────────────────────────────╯ │
│                                                                              │
│ /home/swaayatt7/.local/lib/python3.8/site-packages/kiss_slam/voxel_map.py:42 │
│ in point_cloud                                                               │
│                                                                              │
│   39 │   │   self.map._add_points(vector3fvector)                            │
│   40 │                                                                       │
│   41 │   def point_cloud(self):                                              │
│ ❱ 42 │   │   return np.asarray(self.map._point_cloud()).astype(np.float64)   │
│   43 │                                                                       │
│   44 │   def clear(self):                                                    │
│   45 │   │   self.map._clear()                                               │
│                                                                              │
│ ╭──────────────────────────── locals ────────────────────────────╮           │
│ │ self = <kiss_slam.voxel_map.VoxelMap object at 0x7ff1ffbe0d90> │           │
│ ╰────────────────────────────────────────────────────────────────╯           │
╰──────────────────────────────────────────────────────────────────────────────╯
MemoryError: std::bad_alloc

darshanAtSwaayatt, 2025-04-03:
Hi @saurabh1002 ,

Thanks for the quick reply.

1. It worked without the visualization flag.
2. Without visualization flag RAM consumption was less than 10GB. RAM = 32GB
3. Lidar sensor : vlp32c

saurabh1002, 2025-04-03:
Okay, could you also check the RAM consumption with the visualization flag?

It would tell us if the error was due to excess memory consumption with visualization or something else within our voxel_map implementation. 

If you could share the data with us, we can also test it ourselves. :)

Thanks

saurabh1002, 2025-04-04:
Hey,

We also tested our pipeline again on a dataset with 20,000+ scans from an OS1-128 LiDAR, and we did not run into any errors like yours, even with the visualization flag on. Also, the maximum memory consumption during the run was 12-13 GB of RAM, even with such a dense LiDAR scanner.

[image]

So, the problem you face should be specific to the data you are using. We can only investigate by having access to data or other information about the system's state and visualizer output near the error condition.

saurabh1002, 2025-04-10:
Closing issue due to inactivity. Please feel free to reopen the issue if you need to talk more.

closed 2025-04-10 by saurabh1002