PRBonn/kiss-slam #17: Config for loop closure in indoor environments

radparker, 2025-04-07:
Hey all, many thanks for your contributions. We've been using KISS-ICP on some indoor datasets and it works pretty well but there are certain areas of the indoor environment where error accumulates and the trajectory drifts. I've been experimenting with KISS-SLAM to see if it can detect and close loops to help correct the drift. Do you have some recommendations on a good config for indoor environments? For example, the splitting_distance likely needs to be lowered from 100 meters. When we do that, and retain the same voxel size, for ex 0.15 m, then we do see loop closures happening but there are many false positives. What other values in the config might you suggest that we adjust if we focus on indoor mapping? Thanks!

saurabh1002, 2025-04-08:
Hi @radparker, thanks for using our works. We are working on providing an indoor configuration for public use soon. Until then, I would recommend using the following parameters for the KISS odometry and local-mapping within KISS-SLAM to see if it improves your results.

You can save the config below into a `config.yml` file and pass it to the `kiss_slam_pipeline` with a `--config` option.

```yaml
out_dir: slam_output

odometry:
  data:
    deskew: true
    max_range: 50.0
    min_range: 0.0
  mapping:
    max_points_per_voxel: 20
    voxel_size: 0.5

local_mapper:
  voxel_size: 0.5
  splitting_distance: 5.0
```

Additionally, it would be great if you could share the indoor dataset you are using or at least provide more information about it. This would allow us to investigate and improve our pipeline for even better performance.

radparker, 2025-04-11:
Thanks @saurabh1002 for the advice. Unfortunately, I can't share the particular dataset that I'm looking at right now, but I will let you know if I end up capturing another shareable dataset. I used your config (max_range: 100->50, voxel_sizes: 0.15->0.5 splitting_distance: 100->5) and odometry was unusable (and no loop closures were detected). When I dropped the voxel sizes back down to 0.15, odometry looks good and loop closures are detected, but there are too many false positives.

Below is a result from just running KISS-ICP. The trajectory (blue) looks pretty good until the area circled in red. After, it should follow the green ground truth path but drifts off. I've draw lines in red depicting where potential loop closures could occur.
[image]

When I run the same sequence with KISS-SLAM and the above settings, the odometry looks mostly the same but the false positive loop closures I think lead the graph optimization to warp the graph. Below is the result where I believe each of the detected loop closures is incorrect.
[image]

I'll stay tuned for more guidance around indoor configurations and let you know if I have a shareable dataset that might be useful. In the meantime, if you think of any config changes that I should test, I'd be happy to give it a shot. Thanks!

saurabh1002, 2025-04-12:
It is unfortunate that you cannot share the data.
Another thing you can try maybe is using a higher threshold for the number of inliers required in the loop closure module. This might help reduce the number of false positives.
It is named `self.config.detector.inliers_threshold`.

If the false positives still persists, I would guess that your environment has repetitive structures, in which case I will have to expose some specific paramters from the loop closure module for you to be able to play around with that.