PRBonn/kiss-slam #18: Unable to run on MulRan

ZewdieMc, 2025-04-16:
[image]

saurabh1002, 2025-04-17:
Hi @ZewdieMc 
Are you using a custom-generated rosbag for the MulRAN dataset? 
The default for the MulRAN dataset is `.bin` files in the KITTI format, and there is no such error when I run kiss-slam directly on these files.

saurabh1002, 2025-04-17:
We use the dataloaders from KISS-ICP, and the function to save the result poses in TUM format is also being inherited from KISS-ICP's `OdometryPipeline` class.

[This function](https://github.com/PRBonn/kiss-icp/blob/5701e127e3d9c724cafc46b60d60069464eb4380/python/kiss_icp/pipeline.py#L144) requires a method `get_frames_timestamps` implemented for each specific dataset, as highlighted below.

[image]

At this moment, we have not provided a function `get_frames_timestamps` for the mulran dataloader [here](https://github.com/PRBonn/kiss-icp/blob/main/python/kiss_icp/datasets/mulran.py). As a result you are not able to store the poses in tum format with the timestamps.

To have the timestamps, you can implement a function [similar to the one in kitti dataloader](https://github.com/PRBonn/kiss-icp/blob/5701e127e3d9c724cafc46b60d60069464eb4380/python/kiss_icp/datasets/kitti.py#L88) for the mulran dataloader in KISS-ICP and build it from source.

We will try to update the dataloaders soon, but it might be some time until the changes reflect in the updated pypi install wheels. Feel free to open a PR on KISS-ICP if you do go on to implement the function.

closed 2025-04-18 by saurabh1002

ZewdieMc, 2025-04-18:
Dear @saurabh1002, 
Thank you for the detailed information on which function to work on. I will try to implement the `get_frames_timestamps` method to Mulran.