PRBonn/kiss-slam #23: How to run the pipeline as a Python API Developer ?

independent-researcher-2004, 2025-04-23:
Hi, I set up the code successfully like the instruction in README.md file, but I don't know how to run the pipeline or use the code ? 

Is there anyone can help me with that. 

I tried to run python3 pipeline.py, but it raised an error: 

Running cmake --build & --install in /home/tuandang/tuandang/KISS-SLAM/kiss-slam/build/cp310-cp310-linux_x86_64
Traceback (most recent call last):
  File "/home/tuandang/tuandang/KISS-SLAM/kiss-slam/kiss_slam/pipeline.py", line 34, in <module>
    from kiss_slam.occupancy_mapper import OccupancyGridMapper
  File "/home/tuandang/tuandang/KISS-SLAM/kiss-slam/kiss_slam/occupancy_mapper.py", line 32, in <module>
    from kiss_slam.kiss_slam_pybind import kiss_slam_pybind
  File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1002, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 945, in _find_spec
  File "/home/tuandang/.local/lib/python3.10/site-packages/_kiss_slam_editable.py", line 149, in find_spec
    self.rebuild()
  File "/home/tuandang/.local/lib/python3.10/site-packages/_kiss_slam_editable.py", line 191, in rebuild
    result = subprocess.run(
  File "/usr/lib/python3.10/subprocess.py", line 503, in run
    with Popen(*popenargs, **kwargs) as process:
  File "/usr/lib/python3.10/subprocess.py", line 971, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/usr/lib/python3.10/subprocess.py", line 1863, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: 'cmake'

Thanks a lot!

saurabh1002, 2025-04-23:
Hi @independent-researcher-2004 
If you managed to install the pipeline via `pip install kiss-slam` or via the build from source, then you can simply type `kiss_slam_pipeline --help` to get instructions on how to use it. This is already mentioned in the README.

independent-researcher-2004, 2025-04-23:
Oh thank you. I thought it was only for the pip install kiss_slam part.

Sorry for this confusion. I will try it later.

closed 2025-04-24 by saurabh1002