PRBonn/kiss-slam #24: How to use the kiss_slam_pipeline command ?

independent-researcher-2004, 2025-04-24:
Hi, I am trying to test KISS_SLAM on some datasets, but I don't know how to that, although I ran the command: 

kiss_slam_pipeline --help 

I don't know which datasets the system is supporting. Please show me how to do that, or provide me some instructions to do that. 

Thank you!

benemer, 2025-04-24:
What datasets are you talking about?

independent-researcher-2004, 2025-04-24:
I mean the supported dataset mentioned in the README.md file. 

I don't know what they are and how to run it in particular.

independent-researcher-2004, 2025-04-24:
Oh I just saw the supported dataset in the paper: MulRan [21], HeLiPR [19], and Apollo [17] datasets

I am wondering if I wan to run MulRan dataset, how can I download it and how to run it ? 

Can I download it directly in MulRan's official website ?

benemer, 2025-04-24:
As indicated in the `--help` description, we support all dataloaders from KISS-ICP.

Yes, you can download it from the [website](https://sites.google.com/view/mulran-pr/download).

independent-researcher-2004, 2025-04-24:
Oh thank you, have a nice day sir!

closed 2025-04-24 by benemer