PRBonn/kiss-slam #27: Map not getting generated properly using Ouster 128 channel data

darshanAtSwaayattAI, 2025-05-02:
Hi @saurabh1002 ,

When I tried to generate a map using an Ouster-128 channel lidar(OS2) data, it did not generate properly.
The scanned region is pretty flat, but as we can see in the image below, there is a lot of error along the z-axis.
[image]

Also, a lot of scans are not aligned properly as shown below:
[image]
[image]

Command used: "kiss_slam_pipeline ouster_denoised/"
Are there any specific arguments that I need to enable to generate a proper map?

Thanks and Regards,
Darshan

benemer, 2025-05-12:
Could you give us access to the data? Otherwise, it is hard to tell what is going on.

darshanAtSwaayattAI, 2025-05-14:
Hi @benemer ,

How much data will you need?
Single lidar scan, 10 scans, or all the scans used during the slam process.
The complete data is 25 GB in size.

Also, if you can explain what I can check in the data, what parameters of the data can affect the SLAM process, it will me more helpful.

Also, is there any preprocessing on the raw lidar scans recommended? Like removing any noise?

Thanks and Regards,
Darshan

benemer, 2025-05-14:
A subset of the data in which the z-drift happens is fine. We have observed a drifting odometry in the past, but usually not with dense 128-beam Ouster scans, so that is a bit weird.

The only pre-processing we do is the range clipping and downsampling,  but that all happens internally.