PRBonn/kiss-slam #30: Unable to reproduce paper results

Nussknacker, 2025-05-12:
Hi KISS-SLAM developers,

Thank you for your excellent research and high quality open source projects.

Unfortunately, I'm having trouble reproducing the results of the paper on the Ouster Bridge01 sequence of the HeLiPR data set.
The error is very high, much higher than running KISS-ICP on the same data:

──────────────────────────── Results for HeLiPRDataset Sequence Ouster ─────────────────────────────
+---------------------------------+---------+----------+
|                          Metric |  Value  | Units    |
+=================================+=========+==========+
|       Average Translation Error | 81.614  | %        |
|        Average Rotational Error |  0.178  | deg/m    |
| Absolute Trajectory Error (ATE) | 955.000 | m        |
| Absolute Rotational Error (ARE) |  1.408  | rad      |
|               Average Frequency |   12    | Hz       |
|                 Average Runtime |   80    | ms       |
|        Number of closures found |   97    | closures |
+---------------------------------+---------+----------+
Image output:
[image]

Command used: kiss_slam_pipeline --dataloader helipr --sequence Ouster  /path/to/the/dataset

On the Bridge02 sequence, the ATE is only 21m, so it seems to be related to the sequence.

benemer, 2025-05-13:
The trajectory looks really strange, with many gaps. Can you also post the result of running KISS-ICP? Also, how many scans/poses are you processing here?

Nussknacker, 2025-05-13:
KISS-ICP gives me the following results:
──────────────────────────── Results for HeLiPRDataset Sequence Ouster ─────────────────────────────
+---------------------------------+--------+-------+
|                          Metric | Value  | Units |
+=================================+========+=======+
|       Average Translation Error | 1.402  | %     |
|        Average Rotational Error | 0.003  | deg/m |
| Absolute Trajectory Error (ATE) | 25.098 | m     |
| Absolute Rotational Error (ARE) | 0.056  | rad   |
|               Average Frequency |   14   | Hz    |
|                 Average Runtime |   70   | ms    |
+---------------------------------+--------+-------+

I'm processing the complete Bridge01 Ouster sequence with 21439 scans/poses.

saurabh1002, 2025-05-14:
Hey @Nussknacker ,
We traced this issue back to a new commit on our MapClosures repository (used for loop closure detection).
We have reverted this commit and updated the latest release.
You can go ahead and rerun kiss-slam for the Bridge sequences and let us know if it works fine now.
Thanks again for using our work and highlighting this issue.

Nussknacker, 2025-05-14:
Thank you for your lightning-fast fix. 
I reinstalled kiss-slam and map-closures and used the "--no-cache-dir" option to make sure the packages were re-downloaded. However, it is still not working for me. The history of PyPI also does not show any recent updates: https://pypi.org/project/map-closures/#history.

Do I need to install it from source to get the updated version?

benemer, 2025-05-15:
KISS-SLAM version 0.0.2 is now released and enforces the fixed map-closures 2.0.2. Can you test again on your side if this now fixes your issue?

benemer, 2025-05-15:
I just reran it with the following result:

[image]

Nussknacker, 2025-05-15:
Yes, the new version fixed the issue on my end as well:
 ───────────────────────────────────────────────────── 
                           Metric   Value    Units     
 ───────────────────────────────────────────────────── 
        Average Translation Error   1.625    %         
         Average Rotational Error   0.004    deg/m     
  Absolute Trajectory Error (ATE)   22.558   m         
  Absolute Rotational Error (ARE)   0.060    rad       
                Average Frequency     13     Hz        
                  Average Runtime     77     ms        
         Number of closures found    100     closures  
 ───────────────────────────────────────────────────── 

Thank you very much for the quick fix!

closed 2025-05-15 by Nussknacker