PRBonn/kiss-slam #32: Cannot pip install kiss-slam

timfanselow-bc, 2025-05-15:
On `pip install kiss-slam`, I get the error:

```
ERROR: Could not find a version that satisfies the requirement map_closures>=2.0.1 (from kiss-slam) (from versions: 1.0.1, 2.0.0)
ERROR: No matching distribution found for map_closures>=2.0.1
```

I see in [pyproject.toml](https://github.com/PRBonn/kiss-slam/blob/main/pyproject.toml) the dependency `"map_closures>=2.0.1",`, but on [pypi](https://pypi.org/project/map-closures/#history) I only see releases for `map_closures` 1.0.1 and 2.0.0.

Are you able to push map_closures 2.0.1 to pypi?

benemer, 2025-05-15:
Sorry, we had a problem with the 2.0.1 version. We will release version 2.0.2 today, which also fixes the issues from https://github.com/PRBonn/kiss-slam/issues/30#issuecomment-2880633660.

Sorry for the inconvenience!

benemer, 2025-05-15:
Can you please try again? Version 0.0.2 should now work 👍

timfanselow-bc, 2025-05-15:
Yes looks good, thanks!

I pip installed `kiss-slam`, it installed with no error, fetching `map_closures-2.0.2`

closed 2025-05-15 by timfanselow-bc