PRBonn/kiss-slam #38: Floor points missing in Occupancy Map

mjokesch21, 2025-06-27:
The OccupancyMapper filters the floor points in Occupancy3D-Map. 
Is there a possibility to hold the floor in resulting map?

saurabh1002, 2025-06-27:
@mjokesch21 Please share some more details, maybe some images, the kind of environment, sensor, etc.

mjokesch21, 2025-06-30:
I would like to have the points of the ground (street). In the image you can see that the streets are missing:

[image]

In the local maps the streets are existing:

[image]

mjokesch21, 2025-07-02:
hmm, it is not sufficient. The problem is that Bresenham3DLine generates low occupancy values for the floor points. I guess an additional processing step is necessary after OccupancyMap generation

closed 2025-07-09 by saurabh1002

benemer, 2025-07-09:
I would like to add the following visual explanation: Rays that are close to the ground and have a large incidence angle compared to the ground surface normal will travel through ground voxels, lowering their occupancy value. This is, for example, a problem when using ray-casting for moving object identification, as can be seen in the ERASOR paper:

[image]

Those ground points are considered dynamic because the voxels are marked as free. In your case, this would remove the ground points from the occupancy map. As Gupta suggested, you can try changing the threshold to continue considering those voxels as occupied.