PRBonn/kiss-slam #42: Loop Closures Detected but Pose Graph Optimization Has No Effect (Especially Z Drift in Flat Environments)

alwynmathew, 2025-07-17:
Hi team,

I'm using KISS-SLAM with a VLP-16 to map a ~30x40 m circular road. The odometry works fine, but I’m seeing significant upward drift in the pose graph, resulting in a spiral trajectory instead of a closed loop.

I’ve confirmed the following:

    Loop closures are being detected and accepted — the log shows:

    KissSLAM| Closure Detected  
    KissSLAM| LocalMaps Overlap: 0.89  
    KissSLAM| Closure Accepted  
    KissSLAM| Optimize Pose Graph  
    iteration= 0  chi2= 0.000041  edges= 14

I reduced `local_mapper.splitting_distance` to 20.0 to increase loop candidate frequency (image below). 

I increased `pose_graph_optimizer.max_iterations` to 50.

Still, there is no noticeable correction in the trajectory, and upward drift persists.

Exported `.g2o` trajectory confirms accepted loop closures, but their effect on trajectory is negligible.

 `local_mapper.splitting_distance = 20.0`
[image]
 `local_mapper.splitting_distance = 10.0`
[image]
 `local_mapper.splitting_distance = 10.0`
[image]

Any suggestions? Thanks for your work on this awesome project!

saurabh1002, 2025-07-17:
Hi @alwynmathew 

So there are nuances to consider when dealing with such a sparse sensor and small trajectories.

1. KISS-ICP performs a [double downsampling](https://github.com/PRBonn/kiss-icp/blob/4bd9a1054931bd037ef5e690122832faab621113/python/kiss_icp/kiss_icp.py#L77) of the input point cloud, before the ICP stage. With the [default configs](https://github.com/PRBonn/kiss-icp/blob/main/python/kiss_icp/config/config.py) for KISS-ICP, the resulting downsampled point cloud used for registration will be super sparse for a 16-beam LiDAR, leading to larger drift and curvy trajectory.
2. You did the right thing by reducing the `local_mapper.splitting_distance`  to reduce the size of local maps, however, with the default `max_range` as 100m, each of these local maps will still have a large scanned area within each of these local maps. Even though we exclude closures between three consecutive local maps, your configuration can result in loop closures between local maps within a short range (<40-50m) due to the large overlap in the scene within each local map, since we select the best closure by default in our pipeline which is based on the ORB feature matches. This is also somewhat visible in the trajectory image you posted earlier.
3. This is the reason why the `chi2` error in the pose graph optimization is already quite low, which is why it terminates early, without further iterations.

You can try the following to improve the results:
1. Reduce the `max_range` to `50.0` or even lower; this will already reduce the `voxel_size` to `max_range / 100.0`  as per the default config. This will increase the density of points for registration, reducing drift, while also reducing the size of each local map, avoiding near-range closures.
2. The reduced `local_mapper.splitting_distance` is perfectly fine. You can also include not just the [best closure](https://github.com/PRBonn/kiss-slam/blob/22cd27875234fba343371d0dfd488cd1592dbecf/kiss_slam/loop_closer.py#L45) but also [top-k](https://github.com/PRBonn/MapClosures/blob/b92fefda1d69fe269109915043afd3107c019a6b/python/map_closures/map_closures.py#L44) to consider more loop closure candidates. This will require just an additional for loop to use multiple closures.

saurabh1002, 2025-07-17:
It would be great if you could share the data, so I can also try to figure out a working configuration.

alwynmathew, 2025-07-17:
Thank you for your swift response @saurabh1002 

I have now set 

```
max_range: 50.0
splitting_distance: 20.0
max_iterations: 50
```

console logs:

```
KissSLAM| Closure Detected
KissSLAM| LocalMaps Overlap: 0.4698607242339833
KissSLAM| Closure Accepted
KissSLAM| Optimize Pose Graph
iteration= 0	 chi2= 0.074445	 time= 8.8719e-05	 cumTime= 8.8719e-05	 edges= 7	 Delta= 10000	 step= GN	 tries= 1
iteration= 1	 chi2= 0.000814	 time= 1.0093e-05	 cumTime= 9.88119e-05	 edges= 7	 Delta= 10000	 step= GN	 tries= 1
iteration= 2	 chi2= 0.000814	 time= 8.60495e-06	 cumTime= 0.000107417	 edges= 7	 Delta= 10000	 step= GN	 tries= 1
iteration= 3	 chi2= 0.000814	 time= 8.46386e-06	 cumTime= 0.000115881	 edges= 7	 Delta= 10000	 step= GN	 tries= 1
 27%|█████████████████████████▍                                                                    | 1245/4598 [01:27<05:20, 10.45 frames/s]
KissSLAM| Closure Detected
KissSLAM| LocalMaps Overlap: 0.4309247472856608
KissSLAM| Closure Accepted
KissSLAM| Optimize Pose Graph
iteration= 0	 chi2= 0.003571	 time= 5.22989e-05	 cumTime= 5.22989e-05	 edges= 9	 Delta= 10000	 step= GN	 tries= 1
iteration= 1	 chi2= 0.001966	 time= 1.22159e-05	 cumTime= 6.45148e-05	 edges= 9	 Delta= 10000	 step= GN	 tries= 1
iteration= 2	 chi2= 0.001966	 time= 1.21449e-05	 cumTime= 7.66597e-05	 edges= 9	 Delta= 10000	 step= GN	 tries= 1
 30%|████████████████████████████▎                                                                 | 1386/4598 [01:38<04:07, 13.00 frames/s]
KissSLAM| Closure Detected
KissSLAM| LocalMaps Overlap: 0.5049223924105659
KissSLAM| Closure Accepted
KissSLAM| Optimize Pose Graph
iteration= 0	 chi2= 0.002163	 time= 8.305e-05	 cumTime= 8.305e-05	 edges= 11	 Delta= 10000	 step= GN	 tries= 1
iteration= 1	 chi2= 0.002162	 time= 2.08311e-05	 cumTime= 0.000103881	 edges= 11	 Delta= 10000	 step= GN	 tries= 1
iteration= 2	 chi2= 0.002162	 time= 1.4066e-05	 cumTime= 0.000117947	 edges= 11	 Delta= 5000	 step= GN	 tries= 1
```
No effect on the results

[image]

The whole bag is ~50GB, here is the [first 150sec bag file](https://drive.google.com/file/d/1q7NL1ST0bV-Ve6vFggvxqYGXtW9prcnW/view?usp=sharing). 

My result on the above `first 150sec bag file`:

[image]

saurabh1002, 2025-07-17:
I tried with `max_range = 30.0`  and I get the following output:

[image]

In 3D visualizer for the trajectory, I see very little drift now:

[image]

The individual maps, however, do not look so good because of the errors in KISS-ICP within each local map which arises due to the sparsity of the scanner in use.

alwynmathew, 2025-07-17:
Disable downsampling as mentioned in [Disable dowsmapling ](https://github.com/PRBonn/kiss-icp/issues/128#issuecomment-1505258331) and set

```
data:
  min_range: 0.0

mapping:
  voxel_size: 0.5
```
has improved the map, start and end point of the trajectory is closer.

[image]