PRBonn/kiss-slam #44: Issue with Occupancy Grid maps

srisaripalli, 2025-07-24:
Thank you for the KISS-SLAM project. I am using the following bag file

https://drive.google.com/file/d/1OjanhpXvc047cgo5ovllLQ0f0JFWluVL/view?usp=drive_link

While I can get the trajectory and the map, I cannot get the occupancy 2D map. Here is the config that I am using

out_dir: slam_output

odometry:
    preprocessing:
        max_range: 50.0
        min_range: 0.0
        deskew: true
    registration:
        max_num_iterations: 500
        convergence_criterion: 0.0001
        max_num_threads: 0
    mapping:
        voxel_size: 0.5
        max_points_per_voxel: 20
    adaptive_threshold:
        fixed_threshold: null
        initial_threshold: 2.0
        min_motion_th: 0.1

local_mapper:
    voxel_size: 0.5
    splitting_distance: 5.0

occupancy_mapper:
    free_threshold: 0.2
    occupied_threshold: 0.65
    resolution: 0.5
    max_range: null
    z_min: 2.5
    z_max: 10.0

loop_closer:
    detector:
        density_map_resolution: 0.5
        density_threshold: 0.05
        hamming_distance_threshold: 50
        inliers_threshold: 5
    overlap_threshold: 0.4

pose_graph_optimizer:
    max_iterations: 10

I have tried the default occupancy mapper values also. It looks to me that I am missing something simple since I can't even get a occupancy grid map output i.e. there are no files created. There are only ply files and trajectory files and g2o files created.

saurabh1002, 2025-07-24:
Can you provide the command you ran on the terminal?

srisaripalli, 2025-07-24:
Thank you. I was able to replicate the above. Since I was using a Livox I reduced the max range to 40m. Looks like that was causing issues. Here is a map with max_range 40m and occupancy grid of 0.2 for reference. Will try different splitting distance(s) also since with a splitting distance of 5.0 it is able to find some pose graph optimizations

[image]

saurabh1002, 2025-07-24:
You can try reducing the splitting distance, however, in such a small area the ICP estimates are already good enough. Even if a loop closure is detected, the pose graph chi2 error is quite low already.