PRBonn/kiss-slam #45: Kiss Slam on Sparse Sampled Indoor Data

sendeniz, 2025-07-29:
Dear Kiss Slam Team, 

We are currently trying to use kiss slam for a sparse sampled indoors environment, i.e., the office floor, which includes a corridor and an office. We have collected the point cloud using the following regime: collect a point cloud o - x, y, z every 0.5 meters or if the mobile robot rotates 60 degrees or more. When collection triggers, we also collect the x,y location of the robot and the rotation phi of the wheel, where all 4 wheels have the same rotation. This results in a sample size of 435.

We can visualize the collected point cloud and trajectory by aligning each scan to the robot’s global pose, effectively building a map of where the robot moved and what it saw along the way, which looks like this:

[image]

This looks like an office floor plan. However, when running kiss-slam as followed:
`kiss_slam_pipeline '/home/usr/Downloads/2025_07_01/2025_07_01_drive_0001_sync/livox_points/data' -v`

Pose estimates diverge  and the generated map looks clustered. We get the following result:

[image]

[image]

The data can be found [here](https://drive.google.com/file/d/1f_CPt3jmUBkAX8Y7_8PyVxibmIxuWbcw/view?usp=sharing). 

Note that Kitti and raw kitti works splendidly. There are potential concerns I have with our data collection process, but since its an indoor setting collecting data  points every 100ms seemed a bit redundant at that time. 

Please let me know if you have any questions or answers, I can address regarding this problem setup.

saurabh1002, 2025-07-29:
Hi @sendeniz 
Did you implement your own custom dataloader for the Livox Mid-360 binary pointcloud file structure?
The dataloaders in KISS-SLAM are imported from KISS-ICP, so the default `generic` dataloader will try to parse any `.bin` pointcloud file in the **KITTI** format only.

I would guess that this is the main problem with using KISS-SLAM on your data. Additionally, I can see that the Livox MID-360 user manual shows timestamps as a point cloud data attribute. So I am not sure about your data not having timestamps, although that is not the problem in this case (yet).

saurabh1002, 2025-07-29:
[image]

sendeniz, 2025-07-29:
Hi @saurabh1002,

We've implemented our own data loader and are seeing improved results, particularly in long corridors. However, pose and map divergence still occur in office rooms and during curved motion, as shown below:

[image]

[image]

One possible issue is that we didn’t record per-point timestamps in the LiDAR data. We currently generate synthetic timestamps similar to KITTI, but the data collection team believes using just the full cloud timestamp is sufficient. Do we need to fetch the individual point timestamps ?

Also, does KISS SLAM utilize robot odometry and IMU data (e.g., x, y, z positions and orientation)? Since it works well on KITTI, we assume IMU is used, since KITTI has IMU information but we want to confirm this to help identify the source of divergence in our setup.

saurabh1002, 2025-07-30:
Additionally, a sampling threshold of 60 degrees is too high for a reliable local ICP registration between frames. This will break the point-to-point correspondence model as well as the constant velocity motion prediction model typically used by LiDAR odometry pipelines. You need to either reduce this threshold to 5-10 degrees at least, or try using global point cloud registration techniques that rely on feature correspondences.